nasa/CF #1: Doesn't work by default with cFE6.6

mogar, 2019-11-15:
I'm using the rc-2.2.2 branch with cfe 6.6 and OSAL 4.2.1.0. It doesn't build without the following changes:

1. fsw/src/cf_callbacks.c:1140 - st_size should be FileSize
2. fsw/src/cf_playback.c - all references to os_dirent_t's d_name entry should be changed to FileName.

mogar, 2019-11-15:
Adding needed patch.
[0001-building.patch.txt](https://github.com/nasa/CF/files/3852497/0001-building.patch.txt)

jphickey, 2019-11-16:
The OSAL 5.0.0 version changes these structure names to not conflict/overlap with POSIX-provided terms.  There is also a macro to access the struct members, so one doesn't have to hard-code it.

But I'm not quite following this issue - the change from st_size to FileSize and d_name to FileName would be in the new OSAL, not in version 4.2.1.

mogar, 2019-11-18:
Hmm, I see that now. I had checked out the osal from master, and all of the documentation still refers to 4.2.1, so I assumed that's what I had.

skliper, 2019-11-18:
The included documentation typically just gets updated on release.  See the version notes on the various repos for what's in master (which is the main development branch).  Official releases are under the releases tab on github.  Versions also get reported by cFS at startup.

closed 2020-09-03 by mogar