nasa/CF #4: CF is incompatible with recent OSAL API changes

mbenson1, 2020-06-19:
The cf_playback.c file calls the OS_opendir, OS_readdir, and OS_closedir functions, which were removed in favor of the OS_DirectoryOpen, OS_DirectoryRead, and OS_DirectoryClose functions.

jphickey, 2020-06-19:
Note that OS_opendir, OS_readdir, and OS_closedir were deprecated, not removed.  They still exist for the time being for backward compatibility.  Are you building with OMIT_DEPRECATED set to true by chance?

mbenson1, 2020-06-19:
I see that now.  I will turn that on.  Thanks.

mbenson1, 2020-06-19:
That gets through that issue.  Now I'm running into deprecated data types, i.e. TRUE vs true, and FALSE vs false.  Did you intend to include those in the OMIT_DEPRECATED macro, or are those deleted at this time?

mbenson1, 2020-06-19:
I didn't submit a ticket last night for SCH, FM, and CF because I'm still in the process of porting the code.  So since I'm submitting them now, issues will be coming in piecemeal.

jphickey, 2020-06-19:
Yes, TRUE/FALSE/boolean is also deprecated, in favor of C99 stdbool.h definitions.  However, these should also be in the `#ifndef OSAL_OMIT_DEPRECATED` switch so you'll get them by default.

Are you sure you don't have `OSAL_OMIT_DEPRECATED` defined somehow?  The intent is that by putting these in `#ifndef` as opposed to `#ifdef` that one must opt-in to _not_ get these defined, which should make it backward compatible by default.  I'm surprised if you aren't getting these.  Can you elaborate on your command line and setup/prep procedure?  Maybe there is a documentation issue.

mbenson1, 2020-06-19:
Sorry.  Ran the wrong build in that case.  OSAL_OMIT_DEPRECATED works correctly.

mbenson1, 2020-06-19:
If you decide to bring the hooks and stubs for application testing that was used in previous versions, or if you have some of automated conversion, I can submit tickets covering the unit tests that currently fail.  I have no choice.  I need to still build and run application unit tests, so propagated the hooks and stubs forward.  The FM unit tests build with OSAL_OMIT_DEPRECATED.  I get 854 passes and 43 failures.  

I don't have results from CF.  That's the only application that we do not unit test.

jphickey, 2020-06-19:
I believe a team at GSFC is currently working to update the unit tests for all the CFS apps.  (I'm not directly supporting that effort but AFAIK it is in progress).

Also, AFAIK the next release of the CF is going to be a fairly major update, not just UT but the entire app.  It should be compatible with the latest CFE when it is released.

mbenson1, 2020-06-19:
I heard.  This is good.  We don't unit test CF because for every platform/configuration specific issue we fixed, it would uncover 2 more.  We shelved unit testing on target for later.