nasa/CF #11: Use of Playback Directory command

tyler124, 2021-04-01:
I'd expect the Playback Directory command to function like the following. Assume 7 is the maximum number of transactions that can be occurring simultaneously and that the directory we're attempting to playback has 10 files in it:

- All of the files in the directory get queued.
- Transactions are started for 7 files, at which point the maximum number of transactions is reached.
- Once 1 transaction finishes, CF reads from the queue and starts a transaction for the 8th file.
- Continue until the entire directory has been downlinked.

However, during our testing the command is functioning like the following:
- All of the files in the directory get queued.
- Transactions are successfully started for 7 files, at which point the maximum number of transactions is reached.
- CF attempts to start transactions for the remaining 3 files, but they error out since we're at the maximum number of transactions. The remaining files are never downlinked. 

I'm not sure if we're misunderstanding how the command is intended to be used, or if there may be a bug.

skliper, 2022-05-10:
@astrogeco could this get assigned and closed out?  Would be good to get an answer here and add to user's guide or design material.

skliper, 2022-05-20:
From code inspection of 3.X:
1) Directory gets opened
1) Directory entries are read one-by-one and up to `CF_NUM_TRANSACTIONS_PER_PLAYBACK` files in the directory get queued
1) Based on priority, whatever can get transferred gets transferred
1) Each cycle if any of the files completed it'll queue more directory entries up to the playback limit
1) Continues until entire directory has ben traversed

I haven't tested this yet myself, but it looks like 3.X is close to the expected behavior...

skliper, 2022-05-20:
And confirmed w/ a class 1 transfer of everything in /cf on channel 0:
./cmdUtil -ELE -C3 -A0xB3 --uint8=0 --uint8=1 --uint8=0 --uint8=0 --uint32=0 --string=64:/cf --string=64:bleh

Followed by a bunch of wakeups (could have just added this to the scheduler):
./cmdUtil -ELE -C3 -A0xB5

And observed the start messages (5 immediately started... additional files started after completion to maintain the 5 transaction limit until complete).

dzbaker, 2026-02-26:
May be closeable, needs further verification.