nasa/CF #12: Playback Directory Command - timing bug

tyler124, 2021-06-03:
Related to issue #11 

The playback queue is starting to process a new file after the End-of-File (EOF) PDU is sent for an old transaction, while the old transaction isn’t truly finished (and the ‘active transactions’ count decremented) until the ACK-FIN PDU is received. Due to this, one or more transactions can still be active waiting for their ACK-FIN PDU to be received while the playback queue starts a new transaction. If I had to guess, this is a relic of class1 (unacknowledged) transactions which end after the EOF PDU is sent. 

This issue must have gone unnoticed for so long because in most cases it is hidden, as long as your files are reasonably large and your maximum number of simultaneous transactions is greater than 1 (ours is 7). This is because as long as the file is reasonably large, it will take more time to downlink the new file than it will take for the ACK-FIN from the old transaction to be received and fully closed out. Since the playback queue only transfers one file at a time, you’ll only have a maximum of two transactions active at once.

With maximum number of simultaneous transactions set to 7, if you try to download 8 small files via the playback directory command we encounter the issue (and is the case which found this bug in the first place). This is because the files are so small that 7 separate files can send all of their data and hit the ‘EOF’ phase before the first transaction receives its ‘ACK-FIN’ and fully closes out.

Even if the files are quite large, if I set the maximum number of simultaneous transactions to 1 the issue is encountered every time and the second file will fail to download. This is because a new transaction always attempts to start while the old transaction was between the ‘EOF Sent’ and ‘ACK-FIN received’ phases. 

Would it be valuable for me to submit a bug fix to this repository? Unsure if this repository is monitored.

skliper, 2022-03-24:
Marking as bug for now until v3.X is confirmed to not still have this issue.

astrogeco, 2022-04-22:
@semaldona did you see this in CTF testing?

semaldona, 2022-04-25:
The reported set of conditions was not exercised in the CTF tests.

astrogeco, 2022-05-09:
@semaldona can you check if this is still happening?

semaldona, 2022-05-10:
Confirmed that this issue is not present in the current release of CF 3.0.

Developed a test that issues a class 2 acknowledged directory playback command and confirms that only up to CF_NUM_TRANSACTIONS_PER_PLAYBACK transactions are started from the playback queue.

If CF does not receive an ACK PDU within the table configured ack_timer_s timeout, it will resend the EOF. If an ACK is still not received after the table configured ack_limit retries, CF will close the transaction, issue an error event message, and start the next one in the queue.

skliper, 2022-05-10:
Closing based on confirmation above that behavior no longer exists in CF 3.X.

closed 2022-05-10 by skliper