nasa/CF #17: Cleanup CF v3.0 Perf ID handling

jphickey, 2021-11-23:
**This issue was imported from the GSFC issue tracking system**

_Imported from_: [GSFCCFS-1190] Cleanup CF v3.0 Perf ID handling
_Originally submitted by_: Timmons, Elizabeth J. (GSFC-5820) on Fri Jul 17 13:22:51 2020

_Original Description_:
Encompasses several findings from the CF v3.0 code review:  
  
- Remove unused perf id CF\_DIRREAD\_PERF\_ID  
- Determine whether perf ids CF\_PDU\_RCVD\_PERF\_ID and CF\_VCxPDUSENT\_PERF\_ID should be added (currently used in stakeholder)  
- Ensure that CFE\_ES\_PerfLogExit is called when the app exits

jphickey, 2021-11-23:
Imported comment by internal user on Mon Jul 27 18:22:19 2020

Didn't remove CF\_DIRREAD\_PERF\_ID! It was meant to be used, so I used it!

I can add CF\_PDURCVD\_PERF\_ID. It's just the time it takes to parse a received PDU in the engine.

The CF\_VCxPDUSENT\_PERF\_ID are hard-coded and wrong. Instead there should be a channel PDUSENT perf ID that is defined as a base number with the current channel number being processed used to offset it.

Also stakeholder is currently using it wrong:

1011 if(Chan==0)

{ 1012 CFE\_ES\_PerfLogEntry(CF\_VC1PDUSENT\_PERF\_ID); 1013 CFE\_ES\_PerfLogExit(CF\_VC1PDUSENT\_PERF\_ID); 1014 }else if (Chan==1)

{ 1015 CFE\_ES\_PerfLogEntry(CF\_VC2PDUSENT\_PERF\_ID); 1016 CFE\_ES\_PerfLogExit(CF\_VC2PDUSENT\_PERF\_ID); 1017 }else if (Chan==2)

{ 1018 CFE\_ES\_PerfLogEntry(CF\_VC3PDUSENT\_PERF\_ID); 1019 CFE\_ES\_PerfLogExit(CF\_VC3PDUSENT\_PERF\_ID); 1020 }We could do the same thing for PDURCVD per-channel. Let me know what you think.

jphickey, 2021-11-23:
Imported comment by internal user on Tue Jul 28 09:53:14 2020

I think we should add the PDUSENT and PDURCVD perf IDs

skliper, 2022-03-24:
This issue looks resolved, the PDUSENT, PDURCVD, and DIRREAD perf IDs are being used.

closed 2022-03-24 by skliper