nasa/CF #18: Improve commenting throughout CF v3.0

jphickey, 2021-11-23:
**This issue was imported from the GSFC issue tracking system**

_Imported from_: [GSFCCFS-1194] Improve commenting throughout CF v3.0
_Originally submitted by_: Timmons, Elizabeth J. (GSFC-5820) on Mon Jul 20 13:32:23 2020

_Original Description_:
CF v3.0 needs improved commenting throughout. Specifically:  
  
- file comments in every file describing the purpose of the file  
- explanations and limits for all configuration parameters  
- ensure that all comments are accurate

jphickey, 2021-11-23:
Imported comment by internal user on Mon Jul 27 18:51:30 2020

I've done what's on the list. Do you want me to look for places in the functions to add more comments?

jphickey, 2021-11-23:
Imported comment by internal user on Tue Jul 28 09:52:26 2020

No, as long as you've done what's on the list that should be fine.

jphickey, 2021-11-23:
Imported comment by internal user on Fri Sep  3 08:46:28 2021

This needs reassigned please.

skliper, 2022-03-24:
Related to #206, although this ticket is more general.

PR #294 (2022-07-28): Fix #18 #206 #287, Comment cleanup and spelling
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fix #18 
- Fix #206 
- Fix #287

**Testing performed**
CI

**Expected behavior changes**
None

**System(s) tested on**
CI

**Addition [... cut]
commit 557c1327 (2022-07-28): Merge pull request #294 from skliper/fix206-documentation_1

Fix #18 #206 #287, Comment cleanup and spelling
commit a395ff51 (2022-07-28): Fix #18, Spelling cleanup in comments and events
closed 2022-07-28 by dzbaker