nasa/CF #27: CF method CF_CList_Remove appears to accept bad arguments

jphickey, 2021-11-23:
**This issue was imported from the GSFC issue tracking system**

_Imported from_: [GSFCCFS-1719] CF method CF_CList_Remove appears to accept bad arguments
_Originally submitted by_: Gibson, Alan S. (GSFC-5870) on Thu Aug 26 13:23:38 2021

_Original Description_:
A bad node is passed to CF\_CList\_Remove, but it carries on unaware. This was found because branch 3 of `if((node->next==node)&&(node->prev==node))` can only be covered by a test that passes this situation, node->next == node, node->prev != node.  
  
The issue: should a single node enter, that has a node->next == node (meaning: pointing to itself), but a node->prev != node (meaning: NOT pointing to itself), the code continues as if this is a valid state. It may be impossible for a bad node to enter here, but that is not apparent at the unit test level for CF\_CList\_Remove nor is the Doxygen brief clear on that assumption.

skliper, 2022-06-17:
Note is initialized as node->next = node-prev = node here:
https://github.com/nasa/CF/blob/593d61aaa89a78a31ddabf777ca7176ac540c289/fsw/src/cf_clist.c#L43-L47

Node insert front either keeps as is (first) or updates both:
https://github.com/nasa/CF/blob/593d61aaa89a78a31ddabf777ca7176ac540c289/fsw/src/cf_clist.c#L68-L69

Note insert back either sets both or uses as is (first):
https://github.com/nasa/CF/blob/593d61aaa89a78a31ddabf777ca7176ac540c289/fsw/src/cf_clist.c#L101-L104

Insert after, same pattern:
https://github.com/nasa/CF/blob/593d61aaa89a78a31ddabf777ca7176ac540c289/fsw/src/cf_clist.c#L185-L188

There is no case where only one of next or prev == node possible in the code, since these are internal functions there's no requirement to protect from an impossible case.  Closing as wontfix.

skliper, 2022-06-17:
Actually, makes more sense to me to remove the redundant conditional... just check if next == node which then removes the need for the strange test.

closed 2022-06-22 by astrogeco