nasa/CF #30: CF function CF_CFDP_IsSender(transaction_t *ti) is odd one out because it uses ti

jphickey, 2021-11-23:
**This issue was imported from the GSFC issue tracking system**

_Imported from_: [GSFCCFS-1728] CF function CF_CFDP_IsSender(transaction_t *ti) is odd one out because it uses ti
_Originally submitted by_: Gibson, Alan S. (GSFC-5870) on Tue Sep 14 10:58:59 2021

_Original Description_:
Every other function in cfdp.c that uses a transaction\_t* as an argument names it 't', but CF\_CFDP\_IsSender uses ti.

astrogeco, 2022-04-22:
https://github.com/nasa/CF/blob/96cdf3071996eb98fd1920e7ea39d7af0a2f85c8/fsw/src/cf_cfdp.c#L142-L161

thnkslprpt, 2022-09-19:
> While I do appreciate not wasting characters... neither `ti` or `t` are all that descriptive/helpful in terms of variable names. Maybe `transaction` or `txn` or similar? I'd consider either of these options more "readable".

I would tend to agree with you Jake.
However, there are several hundred uses of all 3 options (t, txn & transaction) in the code base. So for now, it might be best to just correct this particular instance.
For CF_Transaction_t objects specifically, they practically all use ‘t’, so we can align these 2 aberrations to the majority, like Alan Gibson suggested.
Perhaps future work could focus on improving the argument names in general, to make them more readable.

PR #326 (2022-10-03): Fix #30, Correct 2 aberrant instances of CF_Transaction_t argument name
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fixes #30
Amended the only 2 aberrant instances of CF_Transaction_t objects named 'ti', to align them with the  [... cut]
commit a0d061a3 (2022-10-03): Merge pull request #326 from thnkslprpt/patch-1

Fix #30, Correct 2 aberrant instances of CF_Transaction_t argument name
commit a6b47a8d (2022-10-03): Fix #30, Correct 2 aberrant instances of CF_Transaction_t argument name
closed 2022-10-03 by dzbaker