nasa/CF #31: CF function CF_CFDP_ProcessPlaybackDirectory has an oddly bracketed block

jphickey, 2021-11-23:
**This issue was imported from the GSFC issue tracking system**

_Imported from_: [GSFCCFS-1732] CF function CF_CFDP_ProcessPlaybackDirectory has an oddly bracketed block
_Originally submitted by_: Gibson, Alan S. (GSFC-5870) on Wed Sep 22 09:22:05 2021

_Original Description_:
In cfdp.c the function CF\_CFDP\_ProcessPlaybackDirectory has a block of code with no if/while et. Just a bracketed block of code that does not seem to have a reason to be bracketed.

jphickey, 2022-01-06:
The intent of this block may be to scope the local "pt" variable.

Recommend to put variable declarations at the beginning of the function, like coding standards say, then this bracketed block can go away.

astrogeco, 2022-04-22:
Is this the block in question? 
https://github.com/nasa/CF/blob/96cdf3071996eb98fd1920e7ea39d7af0a2f85c8/fsw/src/cf_cfdp.c#L1427-L1463

skliper, 2022-04-25:
Specifically https://github.com/nasa/CF/blob/96cdf3071996eb98fd1920e7ea39d7af0a2f85c8/fsw/src/cf_cfdp.c#L1442-L1462

PR #340 (2022-12-07): Fix #31, reformat bracketed block
**Checklist (Please check before submitting)**

* [X] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [X] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fix #31 

**Testing performed**
Unit testing

**Expected behavior changes**
no impact to behavior

**System(s) tested on**
 - OS: Ubuntu 18.04

**Con [... cut]
commit d8a376b9 (2022-12-07): Fix #31, reformat bracketed block
commit e1daebed (2022-12-07): Merge pull request #340 from havencarlson/fix#31

Fix #31, reformat bracketed block
closed 2022-12-07 by dzbaker