nasa/CF #39: Unreachable code block in check for CF_SEND_NO_MSG

jphickey, 2021-11-23:
**This issue was imported from the GSFC issue tracking system**

_Imported from_: [GSFCCFS-1775] Unreachable code block in check for CF_SEND_NO_MSG
_Originally submitted by_: Hickey, Joseph P. (GSFC-582.0)[VANTAGE SYSTEMS INC] on Tue Nov 16 14:19:04 2021

_Original Description_:
This issue was originally reported by IV&V, creating Jira issue to track its disposition and resolution.  
  
The code within the IF block in cf\_cfdp\_s.c in the CF App source code, namely lines 158-159 will never be executed. The IF block checks variable 'status' if it is CF\_SEND\_NO\_MSG on line 157, this variable originates from the return value of CF\_CFDP\_SendFd(), which can only return values CF\_SEND\_SUCCESS and CF\_SEND\_ERROR.

jphickey, 2022-01-06:
This is still technically true, because CF_CFDP_S_SendFileData gets a message buffer at the very beginning.  The call to CF_CFDP_SendFd in this case is really only sending it, which is different than other Send routines that both acquire a buffer and send it.

Because the buffer was already successfully acquired earlier, checking for CF_SEND_NO_MSG is redundant.  Although technically it is a valid return code from this call per the API, it won't ever be returned in reality.

Probably worth a fix to remove unreachable code, but this is fairly minor.

skliper, 2022-06-17:
OBE - likely went away in one of the refactor commits.  Marking as invalid relative to current implementation.

closed 2022-06-17 by skliper