nasa/CF #57: CF use of "static const int32" for return codes, not used anywhere

jphickey, 2021-11-23:
**This issue was imported from the GSFC issue tracking system**

_Imported from_: [GSFCCFS-1794] CF use of "static const int32" for return codes, not used anywhere
_Originally submitted by_: Hickey, Joseph P. (GSFC-582.0)[VANTAGE SYSTEMS INC] on Wed Nov 17 08:58:47 2021

_Original Description_:
In the table validation function "CF\_ValidateConfigTable" it has return values declared locally as a set of "static const int32" values:  
  
    static const int32 no\_ticks\_per\_second = -1;  
    static const int32 crc\_alignment = -2;  
    static const int32 outgoing\_chunk\_size = -3;  
  
Because they are local, these values are not available to code outside this function, so nothing can ever check for these status codes (and nothing ever does). The validation function caller only ever checks if the value is negative, and CF only checks if the TBL call returns something other than CFE\_SUCCESS.  
  
Since CF is already sending an event with the validation failure details, this error code is not necessary and not relevant. It should be remove and replaced with one of the generic cfe\_error constants.

thnkslprpt, 2022-11-01:
@jphickey which CFE error constants were you thinking might be suitable to substitute in here?
I can't see anything that seems immediately obvious, but maybe I'm not looking in the right place...

jphickey, 2022-11-02:
> @jphickey which CFE error constants were you thinking might be suitable to substitute in here? I can't see anything that seems immediately obvious, but maybe I'm not looking in the right place...

This issue is only about a bit of unneeded/unused complexity.  

The table validation routine(s) are fundamentally boolean in nature, but they use an `int32` status code to be consistent with other CFE API calls.  In that context, `CFE_SUCCESS` means it validated, anything else means it did not validate. 

But CF does not have any recovery or mitigation from a failed table validation.  It may continue to use the old table (not sure) but nothing else is done.

The problem here in CF is that it (potentially) returns 3 different values indicating what failed, but these "constants" are just local to the validation function.  Nothing else checks the return value (aside from CFE_SUCCESS or not), and nothing in CF could even take another alternate action if it wanted to, because the status codes aren't scoped outside of that routine.  In general, returning a symbolic value that is scoped locally within the function is pointless - nothing outside the function can see that symbol, so no caller could check for that return value and handle it if it wanted to.

My recommendation is to just simplify and return a generic CFE error for any of the validation failures.

thnkslprpt, 2022-11-05:
Right that makes sense.
Which CFE error code makes sense though?
Is there actually a generic CFE error or do you mean just something like `CFE_STATUS_EXTERNAL_RESOURCE_FAIL` or `CFE_STATUS_NOT_IMPLEMENTED`?

closed 2023-04-13 by dzbaker