nasa/CF #102: Clean up commented-out functions in unit test

jphickey, 2021-12-08:
The unit test contains a number of test sequences that are completely disabled/commented out.

For example:
https://github.com/nasa/CF/blob/7b99b91cd50a347f8553fc68ea3b074ff0672251/unit-test/cf_cfdp_r_tests.c#L1841-L1876

These serve no value, they are not testing anything, and only serve to clutter the code (it is not likely to work if un-commented, as it would be un-commented already if it did work).

Without a clear reason why these exist in the code, recommendation is to remove.  Version control serves the purpose of preserving historical code, if the concern is to keep a historical record of a previous test case (it does not need to stay in source file).

skliper, 2022-03-24:
https://github.com/nasa/CF/commit/a7f89beac0019073eeaab56ec01ce5476292c7c3 was never merged, reopening since there's still many commented out sections in CF unit tests.

thnkslprpt, 2022-10-01:
> [a7f89be](https://github.com/nasa/CF/commit/a7f89beac0019073eeaab56ec01ce5476292c7c3) was never merged, reopening since there's still many commented out sections in CF unit tests.

Wasn't this all cleared up in 3acc447934793de77d85ced42e28f2d657f816c4 Jake?

skliper, 2022-10-03:
Looks like there's still a few stray lines of commented code that need to get resolved.  Note our coding standard discourages using the C++ style comments to begin with (no `//`).  Might be good to clean that up at the same time.

```
grep -r " //" *
cf_app_tests.c:    // all values for dummy_table.ticks_per_second nominal except 0
cf_app_tests.c:    // all values (except 0) & 3ff == 0 are nominal (1024 byte aligned)
cf_app_tests.c:    // all values less than sizeof(CF_CFDP_PduFileDataContent_t) are nominal
cf_app_tests.c:    // outgoing_file_chunk_size set to greater than sizeof(CF_CFDP_PduFileDataContent_t)
cf_app_tests.c:    // UtAssert_STUB_COUNT(CF_TableInit, 1);
cf_cmd_tests.c:        if (catastrophe_count == 10) // 10 is arbitrary
cf_cmd_tests.c:    // CF_AppData.config_table = &dummy_config_table;
cf_cmd_tests.c:    //  memcpy(CF_AppData.config_table->chan[arg_chan_num].sem_name, dummy_sem_name, 20);
cf_utils_tests.c:    // /* Assert */
cf_utils_tests.c:    uint32 arg_read_size = Any_uint32_LessThan_or_EqualTo(10); // 10 is arbitrary to make test fast
cf_utils_tests.c:    uint8  dummy_buf[10] = {0};                                // 10 to match max read size of 10 (arbitrary)
utilities/cf_test_utils.c:    uint8 diff = ceiling - floor + 1; // +1 for inclusive
```

skliper, 2022-10-03:
Really cf_app_tests.c, cf_cmd_tests.c, cf_timer_tests.c, and cf_utils_tests.c need the scrub all the other test code got.  Removing random numbers, using modern API's, etc.  That's a bigger job, but would finish the other tickets (#264, #114, #105, #87, #86).

thnkslprpt, 2022-10-04:
@skliper so I think 656b2c3e21bc475582a36e6ec7e6aed1e1f4a376 is enough to address this issue in a narrow sense (and close it), and if there are already issues open for the more comprehensive clean-up, the rest can be address there.
Would you agree?

commit a7f89bea (2021-12-09): fix #102, clean out commented code
Removed commented out code in tests since they provided no use
Deleted commented examples of tests
PR #328 (2022-10-18): Fix #102, Clean up remaining commented-out code
**Checklist**
* [x] I reviewed the [Contributing Guide](https://github.com/nasa/osal/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fixes #102.
Remaining stray 'C++ style' comments (single line // comments) in CF were removed to changed to 'C-style' comments ( /* ).
A few remaining lines of comment [... cut]
commit 281a9418 (2022-10-18): Merge pull request #328 from thnkslprpt/fix-102-comment-clean-up

Fix #102, Clean up remaining commented-out code
commit 656b2c3e (2022-10-18): Fix #102, Clean up remaining commented-out code
closed 2022-10-18 by dzbaker