nasa/CF #119: Not all configuration table parameters are validated

semaldona, 2021-12-13:
Per requirement CF7001.1.1, the following values must be validated in the configuration table:
a. Outgoing File Chunk Size 
b. Flight and Ground Entity Ids 
c. Message IDs 
d. CFDP Class 1 or 2

Currently, CF only validates item a. Outgoing File Chunk Size

skliper, 2022-05-20:
There's been some updates (see below), but the table contents need to be reviewed to see if the requirement is currently being met.  Leaving as a requirements issue for now since no elements have been explicitly identified as missing validation.

Requirement was updated to read "... all parameters":
https://github.com/nasa/CF/blob/main/docs/cf_FunctionalRequirements.csv#L94

Currently checks:
ticks_per_second is non-zero
rx_crc_calc_bytes_per_wakeup is nonzero and aligned with 1024
outgoing_file_chunk_size fits in a PDU

Contents are defined in https://github.com/nasa/CF/blob/main/fsw/src/cf_tbldefs.h.
 
Validation code is here:
https://github.com/nasa/CF/blob/893aa3062c55586bc30be3999cf4cec8f7955278/fsw/src/cf_app.c#L118-L149

skliper, 2022-05-20:
Note validating "all parameters" doesn't really make sense for elements that are basically valid for their entire range or similar.  I'd find it easier to read if it just said "...shall validate the configuration table", and the implementation can validate whatever elements are appropriate per a code review (and tested specifically via unit test and generally via functional test).

PR #248 (2022-05-31): Fix #239 #128 #127 #122 #121 #119, Requirements clean-up
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate Contributor License agreement to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fix #239 
- Fix #128 
- Fix #127 
- Fix #122 (note issue #120 still exists)
- Fix #121 
- Fix #119 
- Part #120 (just removes duplicate element from requirement, doesn't fix issue)

Also removed old/obsolete require [... cut]
closed 2022-05-31 by astrogeco