nasa/CF #120: CFDP protocol timer configuration is not per channel

semaldona, 2021-12-13:
Per requirement CF5002, protocol timer configuration should be per channel. Currently the values are applied to all channels.

skliper, 2022-05-20:
None of these are by channel, although the requirements also have issues since 1 and 4 are duplicates.  Either way, these likely should really be per channel since some could be local and others quite far away so changing this to a bug.

CF5002,CF5002,"The following parameters shall be configurable by a CF command:
1. CFDP channel Ack-Timer Value (seconds)
2. CFDP channel NAK-Timer Value (seconds)  
3. CFDP channel Inactivity-Timeout Value (seconds) 
4. CFDP channel Ack-Timer Value (seconds) 
5. CFDP channel Maximum Ack Timeouts (integer counts)
6. CFDP channel Maximum NAK Timeouts (integer counts)","Operational interface to support mission variability and reconfiguration"

CF5002.1,CF5002.1,"The default values for the CFDP channel protocol parameters below shall be defined by a CF Configuration Table:
1. CFDP channel Ack-Timer Value (seconds)
2. CFDP channel NAK-Timer Value (seconds)  
3. CFDP channel Inactivity-Timeout Value (seconds) 
4. CFDP channel Ack-Timer Value (seconds) 
5. CFDP channel Maximum Ack Timeouts (integer counts)
6. CFDP channel Maximum NAK Timeouts (integer counts)",Ensure CF starts in a known and valid operational state

skliper, 2022-05-24:
Note with update in #248, this also fails CF4001.1.1

skliper, 2022-05-24:
Note duplicated element in requirement is addressed in #248, so just the bug remains to resolve.

PR #257 (2022-06-08): Fix #120 #256, Move timers into channel config and remove cfg telemetry
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate Contributor License agreement to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fix #120 
- Fix #256

**Testing performed**
CI (also built and ran unit tests locally)

**Expected behavior changes**
No longer supports reporting config in tlm, get it via table dump (better since it [... cut]
commit 76d54296 (2022-06-08): Fix #120 #256, Move timers into channel config and remove cfg telemetry
commit a7ffdf21 (2022-06-08): Merge pull request #257 from skliper/fix120-time_cfg_per_channel

Fix #120 #256, Move timers into channel config and remove cfg telemetry
closed 2022-06-08 by astrogeco