nasa/CF #154: Incorrect check at the start of CF_Chunks_EraseRange

jphickey, 2022-01-07:
This only verifies that the "start" ID is less than the count:
https://github.com/nasa/CF/blob/b6de2056f70c6d3eb97664a5cd12a50c4978d5bc/fsw/src/cf_chunk.c#L57

It really needs to verify that the end is less than the count too (because it reads this memory).  This could also make the count go negative if the end was beyond the count.

In theory if the code is operating properly this should never happen, but since some error checking is done, it should at least be correctly done.

Recommend to make it check that `(start < end && end <= chunks->count)` instead.

havencarlson, 2022-08-16:
I believe this is no longer necessary
https://github.com/nasa/CF/blob/main/fsw/src/cf_chunk.c#L50-L52

skliper, 2022-08-16:
OBE, specifically #307.  Marking as duplicate and closing.

closed 2022-08-16 by skliper