nasa/CF #192: Trailing zeros for CFDP PDU packets

skliper, 2022-02-04:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I reviewed the [CF README.md file](https://github.com/nasa/CF/blob/main/README.md) to see if the feature is in the major future work.
* [x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
User submitted issue against v3.0 (need to confirm if it still exists in main):
I have discovered that the VCDUs being created on CCSDS virtual channel 1 by the flight software running the CF Task v3 are badly formed.

It appears, probably for performance / debug reasons, that the CF task is limiting a single CFDP PDU per frame. This is a reasonable approach, it is the same thing that we are doing on the DSB SerDes side on virtual channel 2. However, the CCSDS packets being generated only account for the exact length of the contained CFDP PDUs, not the entire remaining contents of the frame. So, in every frame there are at least 7-9 bytes of zeros that are unaccounted for. (Much, much more in the meta data and EoF PDUs.)

Because of the zero bytes trailing the CFDP PDU packets, the decom process in FEDS, ITOS (and probably ASIST) attempt to decom a packet with AppId 0 of length 1 … and get tripped up when they happen to end up inside the following frame which contains a CFDP dump of a series of actual packets recorded on the S/C.

Every transaction on VC01 currently exhibits this behavior. There are literal megabytes of samples captured on the PACE Observatory FEDS. It is very easily reproduced, just initiate a CFDP downlink of any file.

**Describe the solution you'd like**
Size should match packet

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC, project request

jphickey, 2022-02-04:
Just to clarify - when this refers to "virtual channel 1" and "virtual channel 2" are those CFDP channels?  In CF FSW those are numbered 0 and 1, so just want to confirm which channel that refers to -  1-based or 0-based?

I did not observe trailing PDU bytes myself, but that could also be affected by whatever entity is providing CI/TO service as well.

skliper, 2022-02-04:
Yeah, sounds like TO (or similar) is assuming constant size, so not really a CF problem (CF doesn't do any framing).  Maybe CF should have an option to output constant size packets for systems that can't handle variable?

closed 2022-02-08 by elizabeth-wainwright