nasa/CF #206: Add event documentation

skliper, 2022-03-22:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I reviewed the [CF README.md file](https://github.com/nasa/CF/blob/main/README.md) to see if the feature is in the major future work.
* [x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
Event IDs missing user's guide documentation.

**Describe the solution you'd like**
Add descriptions.

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

astrogeco, 2022-05-11:
I'm calling this a "documentation bug"

skliper, 2022-05-11:
I rely on the "bug" label to indicate FSW bugs when reporting in VDDs and communicating w/ stakeholders so I'd prefer not using the general "bug" flag for a documentation issue.  Maybe "critical" or the fact it's also tagged with the Draco milestone could be enough to indicate we need to fix it this round?

astrogeco, 2022-05-11:
Makes sense, removed. I wish we had scoped labels like on gitlab so we could have something like bug::fsw, bug::unit-test, bug::docs

PR #294 (2022-07-28): Fix #18 #206 #287, Comment cleanup and spelling
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fix #18 
- Fix #206 
- Fix #287

**Testing performed**
CI

**Expected behavior changes**
None

**System(s) tested on**
CI

**Addition [... cut]
commit 2ab2341e (2022-07-28): Fix #206, Add command and telemetry documentation
closed 2022-07-28 by dzbaker