nasa/CF #220: Apply latest contributor license agreement links

chillfig, 2022-04-05:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I reviewed the [CF README.md file](https://github.com/nasa/CF/blob/main/README.md) to see if the feature is in the major future work.
* [x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
Changes links from the old contributor license agreement location to its new location.

**Describe the solution you'd like**
Working links

**Additional context**
Depends on https://github.com/nasa/cFS/pull/448

**Requester Info**
Justin Figueroa, ASRC Federal

skliper, 2022-04-05:
Head's up that CF has it's own CLA and shouldn't refer to the CFS CLA.

chillfig, 2022-04-05:
@skliper Ok. Thank you. Where can I find the CF CLA?

PR #255 (2022-06-08): Fix #254, Fix #220, Updates CLA information for Apache 2.0
Adds link to new CLAs in pull_request_template and contributing.md

**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate Contributor License agreement to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fixes #254
- Fixes #220
- Fixes https://github.com/nasa/cFS/issues/495

**Testing performed**
Visually inspected links in fork and in PR template [... cut]
commit 8c2d0405 (2022-06-08): Merge pull request #255 from chillfig/update_app_cla

Fix #254, Fix #220, Updates CLA information for Apache 2.0
commit f6a4fb92 (2022-06-08): Fix #254, Fix #220, Updates CLA information for Apache 2.0

Adds link to new CLAs in pull_request_template,
Edits link language in contributing.md
closed 2022-06-08 by astrogeco