nasa/CF #225: Replace use of `goto`, coding standard violation

skliper, 2022-04-08:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I reviewed the [CF README.md file](https://github.com/nasa/CF/blob/main/README.md) to see if the feature is in the major future work.
* [x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
`goto` violates power of ten and JPL coding standards, typically avoided in cFS code.

**Describe the solution you'd like**
Refactor

**Describe alternatives you've considered**
None

**Additional context**
Static analysis warning that showed up in #218 but wasn't resolved as part of that issue fix

**Requester Info**
Jacob Hageman - NASA/GSFC

astrogeco, 2022-06-21:
@skliper how do you feel about adding this to Draco?

skliper, 2022-06-21:
@astrogeco It's non-trivial, but if we've got the resources it needs to get done.

skliper, 2022-07-13:
Adding to Draco for tracking as @havencarlson has it close to resolved and good to close this violation if there's time this round.  Can always reassess if we run out of time.

PR #269 (2022-07-20): Fix #225, refactor goto statements
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fix #225, replaced all instances of `goto`

**Testing performed**
Ran unit tests

**Expected behavior changes**
No impact to behavior

**System(s) t [... cut]
closed 2022-07-20 by dzbaker