nasa/CF #262: Event ID name doesn't match use

skliper, 2022-06-16:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I reviewed the [CF README.md file](https://github.com/nasa/CF/blob/main/README.md) to see if the feature is in the major future work.
* [x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
`CF_EID_ERR_INIT_CMD_LENGTH` is used for reporting invalid MID received, note `CF_EID_ERR_CMD_GCMD_LEN` and `CF_EID_ERR_CMD_GCMD_CC` are used for actual ground command processing:
https://github.com/nasa/CF/blob/593d61aaa89a78a31ddabf777ca7176ac540c289/fsw/src/cf_app.c#L321-L322

**Describe the solution you'd like**
Event ID names should make sense, maybe `CF_EID_ERR_MID` or similar.

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

thnkslprpt, 2022-10-07:
Maybe `CF_MID_ERR_EID`?
This seems like the most common format in other apps.

Having said that, there is a large variance in formats and order-of-components in names for similar events in other components/apps, so getting to a consistent format at some point in the future would make things easier.

At a glance, there is:
`CF_EID_ERR_INIT_CMD_LENGTH`
`CFE_ES_MID_ERR_EID`
`CFE_EVS_ERR_MSGID_EID`
`CFE_SB_BAD_MSGID_EID`
`CFE_TBL_MID_ERR_EID`
`CFE_TIME_ID_ERR_EID` 
`CS_MID_ERR_EID` 
`HS_MID_ERR_EID`
`MM_MID_ERR_EID` 
`SC_MID_ERR_EID`
`SCH_MD_ERR_EID`

thnkslprpt, 2022-10-17:
ping @havencarlson @skliper

skliper, 2022-10-17:
ping @dmknutsen @dzbaker - they are the deciderers now :)

dmknutsen, 2022-10-17:
Seems very reasonable to go with the most common format - CF_MID_ERR_EID looks good to me...

PR #335 (2024-02-22): Fix #262, Update inaccurate event ID name
**Checklist**
* [x] I reviewed the [Contributing Guide](https://github.com/nasa/osal/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fixes #262
  - Updates Event ID `CF_EID_ERR_INIT_CMD_LENGTH` to a more accurate name and something that more consistent with other apps - `CF_MID_ERR_EID`.

**Testing perf [... cut]
commit 28f39250 (2024-02-22): Fix #262, Update inaccurate event ID name
commit f65d6792 (2024-02-22): Merge pull request #335 from thnkslprpt/fix-262-update-event-id-name

Fix #262, Update inaccurate event ID name
closed 2024-02-22 by dzbaker