nasa/CF #277: Possible silent truncations of entity ID and transaction sequence number

skliper, 2022-07-08:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I reviewed the [CF README.md file](https://github.com/nasa/CF/blob/main/README.md) to see if the feature is in the major future work.
* [x] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

**Is your feature request related to a problem? Please describe.**
The types for entity ID and transaction sequence number are sort of configurable (they are in the config.h but impact packet dfns):
https://github.com/nasa/CF/blob/50c6b04f1980fce3854f8e28cca92bf640612388/fsw/platform_inc/cf_platform_cfg.h#L60
https://github.com/nasa/CF/blob/50c6b04f1980fce3854f8e28cca92bf640612388/fsw/platform_inc/cf_platform_cfg.h#L79

If they are set to anything smaller than uint64 allowed by the spec, then they could get truncated silently when decoding packet headers in CF_CFDP_DecodeHeader:
https://github.com/nasa/CF/blob/50c6b04f1980fce3854f8e28cca92bf640612388/fsw/src/cf_codec.c#L848-L850

Although the impacts don't seem all that critical.  Sequence will roll over locally (should still locally match up), masking off high order eid bits and still getting a match/conflict for destination or source seems unlikely, although possible.

**Describe the solution you'd like**
Check that the encoded value will fit, otherwise send an event and drop the transaction.  Up to 3 extra checks per header decode.  

**Describe alternatives you've considered**
Make these all support the spec allowed 64 bit, no longer configurable.  Would impact memory footprint and cmd/tlm packets.

Leave as is seems like an option also.  Avoids the extra checks on every header that would be needed to avoid something that likely has minimal impact/likelihood.

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

@TheSwarty

acudmore, 2022-08-02:
I can set up the code to check the source_eid, sequence_num, and destination_eid for storage mismatches/silent truncation. 
Is there a preference for generating separate events and error counts for these, or can it be a single check that generates one event and error?
At least have a single error for the source and destination EID since they use the same type and are likely to both have the error at the same time.
The concern is the ability to generate unit tests for the unique errors, especially the two EID size errors.

skliper, 2022-08-02:
I'm fine w/ one event that just reports all three and aborts the transaction if there's truncation... or really whatever is easiest.  Maybe put requested and max supported in the event text?

PR #310 (2022-08-17): Fix #277, Detect pdu truncation
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fixes #277 
Detects and reports silent truncation of Entity ID or Transaction Sequence Number fields.
PDUs with EID/TSN fields that are too big for con [... cut]
commit 6b7e5874 (2022-08-17): Fix #277, reverse new requirement description and rationale
commit 714157b5 (2022-08-17): Fix #277, add requirement
commit 77860544 (2022-08-17): Fix #277, Fix format error and adjust unit test
closed 2022-08-17 by dzbaker