nasa/CF #321: Directory polling does not clean up open directory file descriptors

dennisafa, 2022-08-24:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**
[CF_CFDP_PlaybackDir](https://aetd-git.gsfc.nasa.gov/epg/missions/stp-h9/fsw/cf/-/blob/e6c75da9af2847e3d2e7e50df12455be9a2a8c4b/fsw/src/cf_cfdp.c#L1347) is called by [CF_CFDP_ProcessingPollingDirectories ](https://aetd-git.gsfc.nasa.gov/epg/missions/stp-h9/fsw/cf/-/blob/e6c75da9af2847e3d2e7e50df12455be9a2a8c4b/fsw/src/cf_cfdp.c#L1565) when the interval for the CF_Poll_t structure has expired. This function will attempt to [reopen the directory](https://aetd-git.gsfc.nasa.gov/epg/missions/stp-h9/fsw/cf/-/blob/e6c75da9af2847e3d2e7e50df12455be9a2a8c4b/fsw/src/cf_cfdp.c#L1357). If the polling directory does not continuously receive new files to process, then the OS_DirectoryOpen call happens without a corresponding OS_DirectoryClose call, exhausting the amount of available file descriptors and causing this [error](https://aetd-git.gsfc.nasa.gov/epg/missions/stp-h9/fsw/cf/-/blob/e6c75da9af2847e3d2e7e50df12455be9a2a8c4b/fsw/src/cf_cfdp.c#L1360).
`CF: failed to open playback directory /cf/dl , error=-14`

**To Reproduce**
Steps to reproduce the behavior:
1. Launch CF with a configured polling directory
2. Let ticker expire and attempt to reopen directory
3. Wait until FDs are exhausted and error appears

**Expected behavior**
I expect the directory file descriptor to be cleaned up before each OS_DirectoryOpen call. 

**Code snips**
Configuration table used:
```
CF_ConfigTable_t CF_config_table = {
    10,       /* ticks_per_second */
    33554432, /* max number of bytes per wakeup to calculate r2 recv file crc */
    24,       /* temp local id */
    {{
         32,   /* max number of outgoing messages per wakeup */
         512, /* max number of rx messages per wakeup */
         5,   /* ack timer */
         5,   /* nak timer */
         30,  /* inactivity timer */
         8,   /* ack limit */
         8,   /* nak limit */
         CF_PDU_CMD_MID,
         CF_PDU_TLM_MID,
         512,
         {{1, 0, CF_CFDP_CLASS_1, 21, "/cf/dl", ".", 1}, {0}, {0}, {0}, {0}},
         "", /* throttle sem for channel 1, empty string means no throttle */
         1,
     }},
    1253, /* outgoing_file_chunk_size */
    "/ram",
};
```

**System observed on:**
 - SC3M, Microblaze
 - OS:  Linux 5.4
 - cFS 7.0, CF commit bff67f87

**Additional context**
I added a check in [CF_CFDP_PlaybackDir](https://aetd-git.gsfc.nasa.gov/epg/missions/stp-h9/fsw/cf/-/blob/e6c75da9af2847e3d2e7e50df12455be9a2a8c4b/fsw/src/cf_cfdp.c#L1347) that looks to see if the FD is 0. If it isn't, I close the directory and reopen it. This gets rid of the problem. 

**Reporter Info**
Dennis Afanasev, NASA Goddard Code 587

skliper, 2022-08-24:
I'm not quite following.  The initiate should only happen if the polling directory isn't busy:
https://github.com/nasa/CF/blob/80538827ad62f4a12c91259c98001cfbadd47a87/fsw/src/cf_cfdp.c#L1584-L1596

And busy is only set to zero when the directory has been closed and there's no remaining transactions:
https://github.com/nasa/CF/blob/80538827ad62f4a12c91259c98001cfbadd47a87/fsw/src/cf_cfdp.c#L1499-L1511

Can you elaborate on the flow that causes multiple directory opens without a close?

skliper, 2022-08-24:
Tried to recreate using the main branch and I see a successful close for every open.  Any chance there's something else using up the FDs?