nasa/CF #369: Audit CF for cyclomatic complexity compliance

dmknutsen, 2023-02-28:
Checklist (Please check before submitting)

    [x ] I reviewed the [Contributing Guide](https://github.com/nasa/FM/blob/main/CONTRIBUTING.md).
    [x ] I reviewed the [README file](https://github.com/nasa/FM/blob/main/README.md) to see if the feature is in the major future work.
    [x ] I performed a cursory search to see if the feature request is relevant, not redundant, nor in conflict with other tickets.

Is your feature request related to a problem? Please describe.
NASA guidelines in NPR 7150.2D recommends maintaining a cyclomatic complexity (cc) in software, in particular flight software of 15 or less.

cf_cmd.c:CF_CmdWriteQueue (cc of 18) and cf_cfdp_r.c:CF_CFDP_R_Tick (cc of 17) unnecessarily violate the recommendation of a cyclomatic complexity of 15 or less.

Describe the solution you'd like
Audit cf_cmd.c:CF_CmdWriteQueue and cf_cfdp_r.c:CF_CFDP_R_Tick to see if functions would benefit from simplification such that they comply with NPR 7150.2D recommendation.

Requester Info
Dan Knutsen

chillfig, 2023-06-27:
Will Not Fix. This is the desired implementation at this time. In this case, it would not be possible to re-write while maintaining backwards compatibility and would require re-write of the unit/functional/coverage tests.

skliper, 2023-06-27:
@dmknutsen @chillfig - we may need this for a project, consider reopening?  At minimum it'd be good to be able to trace to an open issue for our documentation.

dmknutsen, 2023-06-27:
That works for me!