nasa/CF #386: CF Draco RC4: Many Fiile Uploads caus CF to Hang or have "Gap Errors"

n-yanchik, 2023-05-23:
**Checklist (Please check before submitting)**

* [X] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [X] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**
On RST WFI FSW, when uploading many files (40-60 files), the CF application either stops responding ang hangs or it responds with 'Gap errors'. This is most easily seen when uploading a large number of table files. 

On RST WFI FSW, this bug can be tracked under https://etdjira.gsfc.nasa.gov/browse/WFI-2531

**To Reproduce**
This bug was originally seen by the WFI Simulator team when trying to upload large numbers of ALD files to the WFI FSW. This bug was reproduced in the WFI FSW Lab COTS GR740 string by uploading 2 to 3 sets of 40 table files. 

From WFI-2531: "The script wfif_fgs_table_dump_and_load_test.prc dumps and reloads all FGS tables (43) total in sequence. Once in a while the script will hang because a table was not able to upload. CFDP page will sometimes show a "gap" error."

**Expected behavior**
The CF application should be able to handle "large" numbers of file uploads without hanging or producing errors. 

**System observed on:**
 - Roman RST WPC, and LEON4 GR740 COTS 
 - RTEMS 5.0
 - CF Draco RC4 (and Draco RC2), WFI FSW 3.2

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Nicholas Yanchik, NASA Goddard Space Flight Center, WFI FSW PDL
nicholas.j.yanchik@nasa.gov

n-yanchik, 2023-08-08:
Upon further investigation, I think this bug can be canceled. This issue seemed to apply to the Draco RC2 release of CFDP, but with the RC4 version of the CF app, the WFI FSW team was able to 'fix' the issue by adjusting/adding delays in our ASIST scripts. 

Note that the ASIST script fixes did not help the issue with the RC2 version of the CF app. There are some more details here: https://etdjira.gsfc.nasa.gov/browse/WFI-2531

dmknutsen, 2023-08-14:
Thanks for the feedback!  We are going to go ahead and close the issue as resolved.

closed 2023-08-14 by dmknutsen

skliper, 2023-08-21:
Numerous gap/chunk bugs were identified in 
- #306

and resolved for Draco-rc3 in
- #307

Not definitively due to this, but if PDU's got dropped and retransmitted these bugs could have caused the problems seen in Draco-rc2.