nasa/CF #392: Seg Fault when invalid destination filename is invalid

the-other-james, 2023-06-27:
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x] I performed a cursory search to see if the bug report is relevant, not redundant, nor in conflict with other tickets.

**Describe the bug**

CF will seg-fault when uploading a file with an invalid destination filename. CF outputs an EVS message that it failed to create the file. It then appears to access a null pointer for a transaction.

Backtrace:
```
EVS Port1 66/1/CF 40: CF: md received for source: aa_put_file_test.txt, dest: cf/cf_test.txt
EVS Port1 66/1/CF 80: CF R0(21:3): failed to create file cf/cf_test.txt for writing, error=-108

Thread 16 "CF" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff7acb700 (LWP 16577)]
CF_CFDP_SendEotPkt (t=0x7ffff7ae9c50 <CF_AppData+1328>) at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_cfdp.c:1698
1698            PktBuf->eot.direction  = t->history->dir;
(gdb) bt
#0  CF_CFDP_SendEotPkt (t=0x7ffff7ae9c50 <CF_AppData+1328>) at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_cfdp.c:1698
#1  0x00007ffff7ad73f2 in CF_CFDP_ResetTransaction (t=0x7ffff7ae9c50 <CF_AppData+1328>, keep_history=0)
    at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_cfdp.c:1607
#2  0x00007ffff7ad5d5e in CF_CFDP_RecvIdle (t=0x7ffff7ae9c50 <CF_AppData+1328>, ph=0x7ffff7ae9a28 <CF_AppData+776>)
    at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_cfdp.c:942
#3  0x00007ffff7adc052 in CF_CFDP_RxStateDispatch (t=0x7ffff7ae9c50 <CF_AppData+1328>, ph=0x7ffff7ae9a28 <CF_AppData+776>,
    dispatch=0x7ffff7ae8760 <state_fns>) at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_cfdp_dispatch.c:197
#4  0x00007ffff7ad48af in CF_CFDP_DispatchRecv (t=0x7ffff7ae9c50 <CF_AppData+1328>, ph=0x7ffff7ae9a28 <CF_AppData+776>)
    at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_cfdp.c:178
#5  0x00007ffff7adbb3c in CF_CFDP_ReceiveMessage (c=0x7ffff7af5ee0 <CF_AppData+51136>)
    at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_cfdp_sbintf.c:295
#6  0x00007ffff7ad7335 in CF_CFDP_CycleEngine () at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_cfdp.c:1575
#7  0x00007ffff7ad3ffe in CF_WakeUp () at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_app.c:278
#8  0x00007ffff7ad4084 in CF_ProcessMsg (msg=0x5555555d9e60 <CFE_SB_Global+14144>)
    at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_app.c:301
#9  0x00007ffff7ad4199 in CF_AppMain () at /sc3m_cfs/cfs_fsw/baseline_apps/cf/fsw/src/cf_app.c:351
#10 0x000055555556bb04 in CFE_ES_TaskEntryPoint () at /sc3m_cfs/cfs_fsw/cfe/modules/es/fsw/src/cfe_es_apps.c:579
#11 0x0000555555596695 in OS_TaskEntryPoint (task_id=65549) at /sc3m_cfs/cfs_fsw/osal/src/os/shared/src/osapi-task.c:138
#12 0x000055555559accb in OS_PthreadTaskEntry (arg=0x1000d) at /sc3m_cfs/cfs_fsw/osal/src/os/posix/src/os-impl-tasks.c:122
#13 0x00007ffff7fa5609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#14 0x00007ffff7eba133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

We are using Cosmos 4.5 with the CFDP engine. We are using the SEND FILE command.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch cFS with the CF app running.
2. Use Cosmos/CFDP engine to send a file, specify an invalid filename (to a directory that doesn't exist, for example)
3. See error in cFS output

**Expected behavior**

CF should report that an error happened (which it currently does), but not seg-fault :)

**Code snips**

None.

**System observed on:**
 - x86_64 PC
 - OS: Ubuntu 20.04
 - Versions: cFE 7.0 rc4, OSAL v6.0.0-rc4, PSP  v1.6.0-rc4 for pc-linux.
 
Note that we have modified cFE, OSAL, and the PSP, but mostly to run on our target system: a microblaze running petalinux (upon which we have also observed the seg-fault).

**Additional context**

Observed with draco-rc2 and draco-rc4

**Reporter Info**
James Marshall, NASA GSFC 587

kylenave, 2023-08-05:
I found the solution to this problem.

In cf_cfdp_r.c near line 600 (I have some changes at the moment):

        t->fd = OS_OBJECT_ID_UNDEFINED; /* just in case */
        if (t->state == CF_TxnState_R2)
        {
            CF_CFDP_R2_SetFinTxnStatus(t, CF_TxnStatus_FILESTORE_REJECTION);
        }
        else
        {
        /// I THINK THIS IS A BUG BECAUSE IT GETS RESET AGAIN AT THE END OF THIS PROCESS AND TRIES TO ACCESS "history" IN THE INTERIM.
       /// The solution is to remove this ... solves the problem for me but I haven't thoroughly explored yet.
            //CF_CFDP_R1_Reset(t);  
        }

dmknutsen, 2023-10-13:
@the-other-james @kylenave  I was able to repeat this error and am investigating further now.  Great catch and thanks for passing this along!!

PR #412 (2023-10-19): Fix #392, seg fault - invalid file destination
**Checklist (Please check before submitting)**

* [x ] I reviewed the [Contributing Guide](https://github.com/nasa/CF/blob/main/CONTRIBUTING.md).
* [x ] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fixes #392 - Adds logic to CF_CFDP_ResetTransaction to check if a transaction has previously been freed prior to resetting the transac [... cut]
commit 046a70ab (2023-10-19): Merge pull request #412 from dmknutsen/Issue_392

Fix #392, seg fault - invalid file destination
commit 0ae687d0 (2023-10-19): Fix #392, seg fault - invalid file destination
closed 2023-10-19 by dzbaker